PrincetonUniversity/ILAng #2: Sharing sub-trees in AST.

Bo-Yuan-Huang, 2017-01-24:
In order to reduce the cost of memory usage, try exporting ASTs without duplicating shared nodes. Add an interface for exporting multiple ASTs at a time.

zhanghongce, 2017-01-24:
I ran the profiling and found it is actually the logging in hash_value() function takes a long time. After removing logging, it ran normally. So I think it is possible to eliminate the shared sub-tree when importing from the file. And I will shortly commit a change about that.

zhanghongce, 2017-01-25:
I added one property to an "abstraction", called **enable_reduce_when_import**, its default value is 0. If it is 1, it will find the shared sub-AST bottom-up during expression import. If it is 2, it will find the shared sub-AST top-down after a complete AST has been loaded.

commit 0435425e (2017-01-27): Close #2. Force sharing when importing a list.
closed 2017-01-27 by Bo-Yuan-Huang