PrincetonUniversity/ILAng #7: Un-driven memory in Verilog export

Bo-Yuan-Huang, 2017-04-12:
Memory variable will be un-driven when there is no any load/store (MEM_next = MEM).

zhanghongce, 2017-04-13:
Because there isn't any store to the mem variable, it doesn't need to be driven in the step or reset condition . 
As with JasperGold, it treats this a little bit differently. JasperGold seems to think a register (or memory) cannot hold its value if it is untouched. You may try adding mem[0] <= mem[0]; in the step condition. 

If it doesn't make JasperGold happy, you can try to write assignment to each address, or even a for loop, which JasperGold can actually unroll. Also make sure it is not blackboxed or it is blackboxed in the proper way you want.

closed 2017-04-13 by Bo-Yuan-Huang

zhanghongce, 2017-04-13:
BTW, the verilog export function has been sligthly changed in risc-v branch. Which allows you to use verilogExpConfig on an abstraction to configure verilog export to make uninterpreted functions as submodules. Later I will merge it to the master branch.