PrincetonUniversity/ILAng #23: Verilog analysis tests fail

Bo-Yuan-Huang, 2018-12-09:
**Describe the bug**
Unit tests for Verilog Analyzer fail. 
``` bash
E1209 16:54:47.826284 106335 verilog_analysis.cc:88] Verilog Analyzer cannot open file: test/t_ana.v
```

**To Reproduce**
Steps to reproduce the behavior:
1. cmake .. -DCMAKE_BUILD_TYPE=Debug
2. make
3. make run_test
4. See error

**Expected behavior**
Test fail with a bunch of error message.
``` bash
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] TestVerilogAnalysis.AnalyzeName
[  FAILED  ] TestVerilogAnalysis.GetTopIo
```

**Screenshots**
``` bash
[0;32m[ RUN      ] [mTestVerilogAnalysis.AnalyzeName
E1209 21:31:38.374091  7592 verilog_analysis.cc:88] Verilog Analyzer cannot open file: test/t_ana.v
E1209 21:31:38.374111  7592 verilog_analysis.cc:41] VerilogAnalyzer cannot proceed because of being in a bad state
E1209 21:31:38.374121  7592 verilog_analysis.cc:41] VerilogAnalyzer cannot proceed because of being in a bad state
/home/travis/build/Bo-Yuan-Huang/ILA-Tools/test/t_verilog_analysis.cc:72: Failure
Expected equality of these values:
  tp_
    Which is: 0
  VerilogInfo::hierarchical_name_type::O_WIRE_w_INTERNAL_DEF
    Which is: 6
/home/travis/build/Bo-Yuan-Huang/ILA-Tools/test/t_verilog_analysis.cc:74: Failure
Value of: VerilogAnalyzerBase::is_io_sig(tp_)
  Actual: false
Expected: true
/home/travis/build/Bo-Yuan-Huang/ILA-Tools/test/t_verilog_analysis.cc:75: Failure
Value of: VerilogAnalyzerBase::is_wire (tp_)
  Actual: false
Expected: true
E1209 21:31:38.374266  7592 verilog_analysis.cc:41] VerilogAnalyzer cannot proceed because of being in a bad state
/home/travis/build/Bo-Yuan-Huang/ILA-Tools/test/t_verilog_analysis.cc:81: Failure
Expected equality of these values:
  tp_
    Which is: 0
  VerilogInfo::hierarchical_name_type::O_WIRE_w_INTERNAL_DEF
    Which is: 6
```

**Additional context**
Starting line 9654 at https://travis-ci.org/Bo-Yuan-Huang/ILA-Tools/builds/465684597

zhanghongce, 2018-12-10:
I'm not able to push to the repo to fix it.

    $ git push
    Counting objects: 10, done.
    Delta compression using up to 8 threads.
    Compressing objects: 100% (10/10), done.
    Writing objects: 100% (10/10), 2.66 KiB | 2.66 MiB/s, done.
    Total 10 (delta 5), reused 0 (delta 0)
    remote: Resolving deltas: 100% (5/5), completed with 4 local objects.
    remote: error: GH006: Protected branch update failed for refs/heads/master.
    remote: error: 5 of 5 required status checks are expected.
    To github.com:Bo-Yuan-Huang/ILA-Tools.git
     ! [remote rejected]   master -> master (protected branch hook declined)
    error: failed to push some refs to 'git@github.com:Bo-Yuan-Huang/ILA-Tools.git'

Bo-Yuan-Huang, 2018-12-10:
The master branch is now protected under status check policy. That is, you should create a branch for the fix/new feature and issue a pull request. Once all checks pass on the pull request, it can then be merged to the master branch.

zhanghongce, 2018-12-10:
Alright.

commit 66564649 (2018-12-29): Fix #23
commit 67a0f7f7 (2018-12-29): Enable vlog analysis test. Fix #23
closed 2018-12-25 by zhanghongce