PrincetonUniversity/ILAng #35: Installation: Why verilogparser's headers are installed?

zhanghongce, 2018-12-18:
**Describe the bug**
1. First I don't think we really need verilogparser's headers to work, because there is a wrapper around it, see `include/ilang/verilog-in/verilog_analysis_wrapper.h`
2. The paths of the verilog parser's include are strange: double include: `include/include`
3. BTW, I'd like to ask (till now I still don't understand), why renaming cpp_api.h to ila++.h, why not just put it as ila++.h, so examples need less modification if we want them to work both before and after installation (just change the path in the makefile)
4. BTW, due to the constant changing of paths of include/library paths and etc. our old examples (e.g. apps/riscv-spec) would not work. Can we keep stable on the code structure (namely, where something is put)
6. BTW, I do recommend using Ubuntu 16.04 and at least GCC 4.9 or above in Travis Cl. 
7. ~~BTW, the pull request check is good that it ensures compilability, but now if I want to keep developing while a pull request is pending, I cannot push to the branch for pull request, as it automatically include all future pushes before pull request is resolved. Okay, one branch for development and one branch for pull request could work. And so will come the merges here and there, if there is not enough already. My recommendation is disable it while in the period of intense development. We don't have that much time wasting on it.~~

**To Reproduce**
- cmake, make, and then make install

**Expected behavior**

**Screenshots**

```
-- Install configuration: ""
-- Installing: /usr/local/lib/cmake/ilang/ilangConfig.cmake
-- Installing: /usr/local/lib/cmake/ilang/ilangConfigVersion.cmake
-- Installing: /usr/local/lib/cmake/ilang/ilangTargets.cmake
-- Installing: /usr/local/lib/cmake/ilang/ilangTargets-noconfig.cmake
-- Installing: /usr/local/lib/libglog.a
-- Installing: /usr/local/include/glog/logging.h
-- Installing: /usr/local/include/glog/raw_logging.h
-- Installing: /usr/local/include/glog/stl_logging.h
-- Installing: /usr/local/include/glog/vlog_is_on.h
-- Installing: /usr/local/include/glog/log_severity.h
-- Installing: /usr/local/lib/cmake/glog/glog-config.cmake
-- Installing: /usr/local/lib/cmake/glog/glog-config-version.cmake
-- Installing: /usr/local/lib/cmake/glog/glog-targets.cmake
-- Installing: /usr/local/lib/cmake/glog/glog-targets-noconfig.cmake
-- Installing: /usr/local/lib/cmake/verilogparser/verilogparserConfig.cmake
-- Installing: /usr/local/lib/cmake/verilogparser/verilogparserConfigVersion.camke
-- Installing: /usr/local/lib/cmake/verilogparser/verilogparserTargets.cmake
-- Installing: /usr/local/lib/cmake/verilogparser/verilogparserTargets-noconfig.cmake
-- Installing: /usr/local/lib/libverilogparser.a
-- Installing: /usr/local/bin/parser
-- Installing: /usr/local/include/include
-- Installing: /usr/local/include/include/verilog_ast_mem.h
-- Installing: /usr/local/include/include/verilog_ast.h
-- Installing: /usr/local/include/include/verilog_parser.h
-- Installing: /usr/local/include/include/verilog_ast_common.h
-- Installing: /usr/local/include/include/verilog_ast_util.h
-- Installing: /usr/local/include/include/verilog_preprocessor.h
-- Installing: /usr/local/lib/cmake/ilasynth/ilasynthConfig.cmake
-- Installing: /usr/local/lib/cmake/ilasynth/ilasynthConfigVersion.cmake
-- Installing: /usr/local/lib/cmake/ilasynth/ilasynthTargets.cmake
-- Installing: /usr/local/lib/cmake/ilasynth/ilasynthTargets-noconfig.cmake
-- Installing: /usr/local/lib/libilasynthAst.a
-- Installing: /usr/local/lib/libilasynth.a
-- Installing: /usr/local/include/ilasynth
-- Installing: /usr/local/include/ilasynth/ast
-- Installing: /usr/local/include/ilasynth/ast/bitvec.hpp
-- Installing: /usr/local/include/ilasynth/ast/bvinrange.hpp
-- Installing: /usr/local/include/ilasynth/ast/bool.hpp
-- Installing: /usr/local/include/ilasynth/ast/mem.hpp
-- Installing: /usr/local/include/ilasynth/ast/func.hpp
-- Installing: /usr/local/include/ilasynth/ast/hash.hpp
-- Installing: /usr/local/include/ilasynth/ast/choice.hpp
-- Installing: /usr/local/include/ilasynth/ast/node.hpp
-- Installing: /usr/local/include/ilasynth/EqvChecker.hpp
-- Installing: /usr/local/include/ilasynth/ast.hpp
-- Installing: /usr/local/include/ilasynth/exception.hpp
-- Installing: /usr/local/include/ilasynth/simplify.hpp
-- Installing: /usr/local/include/ilasynth/MicroUnroller.hpp
-- Installing: /usr/local/include/ilasynth/smt.hpp
-- Installing: /usr/local/include/ilasynth/common.hpp
-- Installing: /usr/local/include/ilasynth/util.hpp
-- Installing: /usr/local/include/ilasynth/horn.hpp
-- Installing: /usr/local/include/ilasynth/boogie.hpp
-- Installing: /usr/local/include/ilasynth/abstraction.hpp
-- Installing: /usr/local/include/ilasynth/funcReduct.hpp
-- Installing: /usr/local/include/ilasynth/synrewriter.hpp
-- Installing: /usr/local/include/ilasynth/logging.hpp
-- Installing: /usr/local/include/ilasynth/VerilogExport.hpp
-- Installing: /usr/local/include/ilasynth/Unroller.hpp
-- Installing: /usr/local/include/ilasynth/assert.hpp
-- Installing: /usr/local/include/ilasynth/imexport.hpp
-- Installing: /usr/local/include/ilasynth/rewriter.hpp
-- Installing: /usr/local/include/ilasynth/cExport.hpp
-- Installing: /usr/local/include/ilasynth/cppsimgen.hpp
-- Installing: /usr/local/include/ilasynth/memvalues.hpp
-- Installing: /usr/local/include/ilasynth/type.hpp
-- Installing: /usr/local/include/ilasynth/exportSMT.hpp
-- Installing: /usr/local/include/ilasynth/synthesizer.hpp
-- Installing: /usr/local/include/ilasynth/genCBMC.hpp
-- Installing: /usr/local/lib/libilang.a
-- Installing: /usr/local/include/ila++.h
```

**Desktop (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - cmake [3.10.2]

Bo-Yuan-Huang, 2018-12-19:
1 and 2 are already in the TODO list ([issue 3](https://github.com/Bo-Yuan-Huang/verilog-parser/issues/3)).
3. We should always use ila++.h in the example. But since changing it is simple, we can do it then. 
4. Yes, we need to migrate previous examples as the simple example provided ([simple](https://github.com/Bo-Yuan-Huang/ILA-Tools/tree/master/examples/simple)). 
5. I can add another test matrix using 16.04 (this would increase build time due to updating CMake). 
6. Headers other than ilang++.h are not installed by default (not for users). However, you can turn it on with -DILANG_INSTALL_DEV=ON
7. We can disable it for now due to the intense deadline. However, it is dangerous since I've been spending time debugging for issues that could be found by the CI tests but did not due to the merge with an unchecked branch.

Bo-Yuan-Huang, 2018-12-20:
Actually, I don't get why 3 is a problem. Including ``cpp_api.h`` should never work even before installation, unless you put it as part of the library..., which you should never do for an example.

closed 2018-12-20 by Bo-Yuan-Huang

zhanghongce, 2018-12-21:
Then for 7, I will merge to the master branch on my side and use it for pull request.

PR #39 (2018-12-20): Travis. 
Fix #35, #36, #37, #38.