PrincetonUniversity/ILAng #68: Hash Conflicts are More Frequent than Expected

zhanghongce, 2019-01-23:
This issue is related to the hash function used in `src/ila/hash_ast.cc`

The test is provide in this [pull request](https://github.com/Bo-Yuan-Huang/ILAng/pull/67) (but I don't expect you to merge it, because the test will most likely fail).

In the test constructed, in each round there are 

 - 512x ITE nodes
 - 512x EQ nodes
 - 1x BitVectorVar node
 - 1024x BvConst nodes
 
The BvConst nodes are the same for each round, so they should not result in different
hashes, and will be replaced by the same BvConst node as the first round.
So in the `i`-th round, there will be `1025*i + 1024` AST nodes (assuming `i` starts from 0)

So in 10000 rounds, there will be around 10^7 AST nodes.

According to table [here](https://preshing.com/20110504/hash-collision-probabilities/), the collision probability for a 64-bit hash with 1.9x10^7 hash values is 10^-5. However, I was never able to finish this test without errors (you can also try it yourself). If it is not a hash-function design problem, I think we should go to buy lottery.

zhanghongce, 2019-01-23:
The test results on Travis CI are also very mysterious....

Bo-Yuan-Huang, 2019-02-21:
Address in ca55d46f51e6ec43d7e110f3032b6798c27d4c06

closed 2019-02-21 by Bo-Yuan-Huang