PrincetonUniversity/ILAng #83: MCM axiom helper build fail under release mode

Bo-Yuan-Huang, 2019-04-27:
**Describe the bug**
Build fail under Release model.

**To Reproduce**
Steps to reproduce the behavior:
1. cmake .. -DCMAKE_BUILD_TYPE=Release
2. make
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

``` bash
ILAng/src/mcm/axiom_helper.cc: In member function ‘z3::expr ilang::MemoryModel::SameAddress(const ilang::TraceStep&, const ilang::TraceStep&, const string&, ilang::MemoryModel::AxiomFuncHint, ilang::MemoryModel::AxiomFuncHint) const’:
ILAng/src/mcm/axiom_helper.cc:414:30: error: invalid use of void expression wexpr->Print(ILA_INFO); 
```

**Environment (please complete the following information):**

-   OS: Ubuntu 16.04
-   Compiler gcc 5.4.0

zhanghongce, 2019-04-28:
Actually, I'm curious, what will `ILA_INFO` become under Release mode?

Bo-Yuan-Huang, 2019-04-28:
INFO, WARN, ERROR, ASSERT, and DLOG are Debug mode only:
https://github.com/Bo-Yuan-Huang/ILAng/blob/61f90d3b6062a2c3d37edc95503eac949b5c6534/include/ilang/util/log.h#L13-L32

For critical assertions, use CHECK instead, which persists in both Debug and Release mode:
https://github.com/Bo-Yuan-Huang/ILAng/blob/61f90d3b6062a2c3d37edc95503eac949b5c6534/include/ilang/util/log.h#L34-L47

Bo-Yuan-Huang, 2019-04-28:
This is the magic of Glog so that you can use `<<` in both modes without breaking the build system.

commit a16935bb (2019-04-28): Merge pull request #84 from zhanghongce/master

close #82 close #83. Fix using ILA_* as stream objects.
closed 2019-04-28 by Bo-Yuan-Huang