PrincetonUniversity/ILAng #114: False alarm in back-to-back file generation

Bo-Yuan-Huang, 2019-06-07:
**Describe the bug**
When creating an output directory (e.g., in Verilog target generation) back-to-back, there would be an error message saying directory creating fails. However, the directory already exists and the file generation is Okay. 

**To Reproduce**
Steps to reproduce the behavior:
1. Generate Verilog target
2. Generate Verilog target again -- with the same output path
3. See error

**Environment (please complete the following information):**

-   OS: Ubuntu 18.04
-   Compiler gcc 7.4.0

**Additional context**
Check status code to be -1 instead of 0.

zhanghongce, 2019-06-08:
Do you see an error or a warning?

Currently it should be something like this:
```
W0607 21:13:49.464702 10027 single_target.cc:1093] Cannot create output directory:../verification/XX
```

If this is what you see, I don't see why this is a bug. The user should be prompted that of such issues in case there are other potential problems (directory accessing privilege is ill-set).

yuex1994, 2019-06-08:
I guess the "bug" means the user needs to handle the warning in his program but not let it be printed out.

Bo-Yuan-Huang, 2019-06-08:
I gave it a bug tag just because there's no better tag for this... A better solution is to make the warning message clearer -- differentiate already-exist from create-fail (which can be due to privilege issue). 

No worry, I've changed that in a branch and will PR soon.

PR #115 (2019-06-08): Is114
fix #114
closed 2019-06-08 by Bo-Yuan-Huang