PrincetonUniversity/ILAng #126: Adapt vtarget-cosa generation to cosa's new interface

yuex1994, 2019-06-22:
**Describe the bug**
Some of Cosa's interfaces are changed. Vtarget-gen cosa needs to adapt to it.

Reference change in Cosa: https://github.com/cristian-mattarei/CoSA/commit/c06cd9ac6a0cc6bb81ccb76d2bd64e9df58b4e6b#diff-9bf6f442ae6e7232f67703a17a9385b6

**To Reproduce**
Steps to reproduce the behavior:
1. Run the AES demo with lastest Cosa and ILAng
2. See error

zhanghongce, 2019-07-08:
The change to CoSA interface is trivial, however I'm having troubles running the latest version of CoSA...
See also: https://github.com/cristian-mattarei/CoSA/issues/100

zhanghongce, 2019-07-12:
Hi, I have a question about the maintenance plan... When do we plan to switch to use new CoSA? I'm concerned that (1) the new problem file for CoSA is incompatible with the old one (2) for new CoSA, it seems that it requires a newer version of Yosys.

zhanghongce, 2019-07-13:
See also: https://github.com/Bo-Yuan-Huang/ILAng/pull/131

commit 327fdba3 (2019-07-18): Merge pull request #131 from zhanghongce/master

fix #126
closed 2019-07-18 by Bo-Yuan-Huang