PrincetonUniversity/ILAng #154: Extend constant bit-width

Bo-Yuan-Huang, 2019-10-30:
**Describe your feature request.**
32-bit is too less for signed-constant in many applications.

**Describe the solution you'd like**
Change the internal variable data type from int to, for example, int64_t. 

**Additional context**
Test with nibbler wrapper generation and LMAC generation.

zhanghongce, 2019-10-31:
One more thing on this, the current verilog-generation will need modification to handle this widening. 
Please ping me if the changes are going to be merged...

zhanghongce, 2019-10-31:
Also, Huaixi notified me another issue in verilog-generation, about handling negative numbers (this is sometimes unavoidable, as BvConst uses the `signed` int for storage, even if someone writes in 2's complement when using the APIs, it will be converted to a negative number internally if it is wide enough) I'm not sure what's the opinion on using the `signed` number there, but I'm planning to add the conversion to 2's complement in verilog-generation.

Bo-Yuan-Huang, 2019-10-31:
The change will be made in branch/PR #156 
Currently, I'm changing bit-vector values to `int64_t`, but can be easily changed to `uint64_t` or `size_t`.
Seems like most of use cases don't need the `signed` feature, shall we change it to the `unsigned` version?

zhanghongce, 2019-10-31:
It would be slightly easier for the verilog generator to handle just the unsigned case. But one more question, is there a range check in `ilang::BvConst`?

Bo-Yuan-Huang, 2019-10-31:
Then let's change to the unsigned version. 
No checking now - can add it as well

zhanghongce, 2019-10-31:
Okay, thanks!

zhanghongce, 2019-10-31:
BTW, would you like me to commit to `issue-154` or start a new branch for updating verilog-gen?

Bo-Yuan-Huang, 2019-10-31:
Same branch may be better, since they are correlated?

zhanghongce, 2019-10-31:
Okay, no problem.

PR #156 (2019-11-01): Issue 154
Fix #154, #155. 
- change numeric type from `signed int` to `uint64_t`
- support legacy z3 APIs for different numeric types
- add range check when defining bit-vector constant
closed 2019-11-01 by Bo-Yuan-Huang