PrincetonUniversity/ILAng #178: Seg fault in VlgTargetGen unit tests

Bo-Yuan-Huang, 2020-06-02:
**Describe the bug**
The unit test `TestVlgTargetGen.MemoryInternal` fails due to segmentation fault.

**To Reproduce**

1. Configure and build the default setting.
2. run `./test/unit_test --gtest_filter=TestVlgTargetGen.MemoryInternal`
3. See error

**Environment:**

-   OS: Ubuntu 18.04
-   Compiler gcc-8

**Additional context**
Found by the Semaphore 2 CI. https://ilang.semaphoreci.com/jobs/c330fed9-5665-4473-9d42-08005c3c147a#

zhanghongce, 2020-06-07:
@Bo-Yuan-Huang I'm afraid that I'm having difficulty to reproduce this. Are you sure it is using gcc-8? Line 866 of the log says: `-- The CXX compiler identification is GNU 4.8.5`

Bo-Yuan-Huang, 2020-06-07:
Thanks for catching this - maybe it's due to the inconsistent setting. Let me test it and get back later.

closed 2020-06-07 by Bo-Yuan-Huang