PrincetonUniversity/ILAng #179: Failure to concatenate constant with specified length in systemc model generation

LeeOHzzZ, 2020-06-02:
**Describe the bug**
In ILA model, concatenation with specified length of constant is often used to expand the variable's length. However, the generated systemc model fails to concatenate the specified length of constant.
e.x. 
ILA code: var1 = Concat(var2, BvConst(0,4)), we should expected the var1 in systemc has a bitwidth of len(var2) + 4. However, the generated systemc code is like this:
SystemC code: var1_sc = (var2_sc, 0), which only has bitwidth of len(var2_sc) + 1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ILA model and specify a variable of var1 = Concat(var2, BvConst(0, 4))
2. Build the ILA model and generate the systemc codes
3. Look into the var1 update function (decode_*****), and you can see the error

**Environment (please complete the following information):**

-   OS: [e.g. Ubuntu 18.04]
-   Compiler gcc8

yuex1994, 2020-06-07:
Need more details (e.g. specific model) to reproduce the bug. 

Provided example (attached) doesn't expose bugs:
    **auto s1 = m.NewBvState("s1", 4);
    auto s2 = m.NewBvState("s2", 8);
    auto c1 = BvConst(0, 4);
    auto c2 = BvConst(0, 8);
    auto instr1 = m.NewInstr("i1");
    instr1.SetUpdate(s2, c2 + Concat(s1, c1));**
In the generated file, the intermediate variable for Concat(s1, c1) and c2 + Concat(s1, c1)) both have length 8.

Attachment:
[example.zip](https://github.com/Bo-Yuan-Huang/ILAng/files/4741291/example.zip)

LeeOHzzZ, 2020-06-07:
@yuex1994 
The bitwidth may be correct, but the value is wrong for the intermediate result.
For example, if s1 == BvConst(1,0), and s2 = Concat(s1, BvConst(0, 4)), then s2 should be equal to 16. However, in the current version, in systemc model, s2 = (s1,0), which is equal to 2 instead, which means it only append 1 bit after s1 for the concatenation.

yuex1994, 2020-06-08:
I see. Fix -- [pr#186](https://github.com/Bo-Yuan-Huang/ILAng/pull/186)