PrincetonUniversity/ILAng #216: Fix the link to API reference

zhanghongce, 2021-05-10:
**Describe the bug**
The link to the API reference in README.md is broken after repo migration, namely the link `https://bo-yuan-huang.github.io/ILAng-Doc/doxygen-output-html/namespaceilang.html` is no longer valid.

**To Reproduce**
Steps to reproduce the behavior:
1. Click the `docs:doxygen` badge in README.md

Discussion: ILAng-Doc is not hosted under Bo-Yuan's name. However, I don't know if the university has a corresponding `github.io` link for use.

yuex1994, 2021-05-10:
Thanks for noticing this -- https://github.com/PrincetonUniversity/ILAng/pull/212 should fix this, but it has not been merged yet.

zhanghongce, 2021-05-10:
Thanks, Yue!

closed 2021-05-10 by zhanghongce